wpilibsuite/frc-characterization #3: Add other types of characterization

virtuald, 2018-05-05:
I really like this methodology for controlling the robot over networktables... it allows the hard logic to be written once for all languages, and simple code can be ran on the robot side. There are probably other useful ways we can use this methodology to help teams control their robot better.

One thing I'm very interested in is measuring information that would allow us to simulate drivetrain turning better.

Other things that could be done are automated programs to compute feed-forward such as in the TalonSRX user manual.

Perhaps utilities for tuning a robot too?

Oblarg, 2019-05-17:
I've just made an arm characterization branch.  Perhaps someone with a robot handy could test it?

virtuald, 2019-05-18:
It looks cool. Unfortunately, I don't have a robot available to me.

closed 2019-10-16 by Oblarg