wpilibsuite/frc-characterization #32: Docs are incorrect about python version

clrozeboom, 2019-10-20:
I tried using this with Python 3.6 and ran into an issue with importlib.resources not being found.
I then updated to Python 3.8 and had problems installing robotpy-characterization.
3.7 seems to be the Goldilocks version, though the docs say 3.6 and greater.

Oblarg, 2019-10-20:
Yes, 3.8 is broken, I discovered this independently last night.  I'll update the dependency list and docs accordingly.

calcmogul, 2019-10-20:
Numpy hasn't made a release for 3.8 yet, so that's why it won't install.

Kevin-OConnor, 2019-11-11:
Numpy has released for 3.8 put matplotlib has not. Seems like the readmes should be updated to suggest exactly 3.7 until matplotlib 3.2 is released. The PyPI package specifies 3.7 so the current readme note about 3.6 definitely needs to be updated.

Kevin-OConnor, 2019-11-11:
Looks like statsmodels also doesn't have a release that works on 3.8 yet. This issue suggests that version 11 will likely support 3.8: https://github.com/statsmodels/statsmodels/issues/6220 The described workaround there did not work for me to get statsmodels running on 3.8

Starlight220, 2020-09-16:
The frc-docs articles about this state clearly that only Python 3.7 should be used. So this has been resolved and can be closed. If someone wants to check if the dependency now has a 3.8 version, go for it!

BobSaidHi, 2021-04-15:
Both `numpy `and `statsmodels `seem to support Python 3.9 now.  Are there any plans to support Python 3.9?
I could help test anything that needs testing but I don't have access to a robot right now.

Piphi5, 2021-04-15:
This tool will be replaced with https://github.com/wpilibsuite/sysid for the 2022 season so there probably isn't a need to update it for Python 3.9 support.

calcmogul, 2021-04-15:
It probably wouldn't be hard, but frc-characterization has not unit tests or integration tests, so making sure we didn't break something more is difficult.