wpilibsuite/frc-characterization #46: Invisible button labels on macOS dark mode

RobinsonZ, 2019-10-29:
On macOS, when system-wide dark mode is enabled, all button labels have invisible text as seen here:

[image]

When buttons are clicked, the text is visible in white:
[image]

With dark mode disabled, the buttons appear normal.

Oblarg, 2019-10-29:
Unfixable.  This is a bug with tk on mac.

Oblarg, 2019-10-29:
https://stackoverflow.com/questions/52529403/button-text-of-tkinter-does-not-work-in-mojave

pietroglyph, 2019-12-25:
This appears to be fixed on Catalina. We should still leave this issue open, because Mojave is not unsupported.