wpilibsuite/frc-characterization #57: Regarding "Do not omit trailing commas..."

virtuald, 2019-11-12:
On the readme:

> All list values must be lists, even if they have only one element. Do not omit trailing commas (e.g. in (False,)), as this will cause them not to be parsed as a list

Strictly speaking, things surrounded with parentheses are tuples, lists are surrounded by brackets. I tend to use tuples to denote things as immutable, but often lists and tuples are interchangeable.

Why does this matter? Well, `[False]` is a valid list even without a trailing comma, and so it's a lot harder to mess up. I recommend changing the documentation to use lists (brackets) instead of tuples (parentheses).

Oblarg, 2019-11-12:
We already made this change in the code; we just haven't updated the documentation to match.

Starlight220, 2020-09-16:
Since the documentation is in frc-docs, shouldn't this issue be renamed (as it would lack some context) and moved there? Assuming that this hasn't been fixed by the docs for this moving into frc-docs.