wpilibsuite/frc-characterization #70: Only use the needed vendor libraries

Daltz333, 2019-12-31:
Generated projects should only use the vendordeps that it needs. It's not really feasible to expect every team to have vendor setups installed locally for things they don't use. Deleting the offending vendordep is a workaround, but not a solution

Oblarg, 2019-12-31:
The vendordeps should auto-install, so this shouldn't be necessary; it still might be an okay improvement, I'm unsure if it's worth the extra duplication.

swalberg, 2020-02-08:
FYI we're using an ADIS16448 so we added the JSON to vendordeps. Some library being brought in by `navx_frc.json` and `REVRobotics.json` was preventing the ADIS from being discovered ("can not find ADIS16448"). Deleted the superfluous JSON, rebuilt, and it's working.