wpilibsuite/frc-characterization #72: Evaluate whether other PigeonIMU methods should be used

auscompgeek, 2020-01-01:
I suspect `getFusedHeading()` would be better for getting the current heading, for example.

pietroglyph, 2020-01-03:
If I remember correctly (this is not mentioned on the Phoenix javadoc), `getFusedHeading()` returns an angle modded by 360. We need a continuous angle, because we need to know how many revolutions have been made when we determine track width.

We could detect rollovers from 360 deg to 0 deg if you think that the additional accuracy is important, but I didn't have any issues last year using `getRawHeading()` to get the track width for trajectory following on our competition robot.

auscompgeek, 2020-01-04:
Ah, fair enough. Just that currently the generated project allocates a 3-array and reads the Z angle from that…

What other methods does the PigeonIMU have that returns the yaw?

pietroglyph, 2020-01-07:
There's `getYawPitchRoll` and `get6dQuaternion`. The latter will be normalized on [-180, 180] when you convert to Euler angles, because you have an `atan2` in that process. I don't know if `getYawPitchRoll` returns a non-modded angle. I vaguely recall it being modded from when we used it last year.