wpilibsuite/frc-characterization #88: Diagnostics Crash on MacOS

randomstring, 2020-01-18:
I have characterization data that I'm trying to analyze on my Mac. The **Analyze Data** works fine, however click any of the **Diagnostic** buttons crashes the app. Here is a sample stack trace

`2020-01-18 13:49:26.787 python3.7[52433:6406129] WARNING: <NSOpenPanel: 0x7fc19cb068c0> running implicitly; please run panels using NSSavePanel rather than NSApplication.
Fatal Python error: PyEval_RestoreThread: NULL tstate

Current thread 0x000000010aca5dc0 (most recent call first):
  File "/Users/dole/.virtualenvs/py3/lib/python3.7/site-packages/matplotlib/backends/backend_macosx.py", line 193 in mainloop
  File "/Users/dole/.virtualenvs/py3/lib/python3.7/site-packages/matplotlib/backend_bases.py", line 3309 in show
  File "/Users/dole/.virtualenvs/py3/lib/python3.7/site-packages/matplotlib/cbook/deprecation.py", line 413 in wrapper
  File "/Users/dole/.virtualenvs/py3/lib/python3.7/site-packages/matplotlib/pyplot.py", line 269 in show
  File "/Users/dole/.virtualenvs/py3/lib/python3.7/site-packages/frc_characterization/drive_characterization/data_analyzer.py", line 1173 in _plotVoltageDomain
  File "/Users/dole/.virtualenvs/py3/lib/python3.7/site-packages/frc_characterization/drive_characterization/data_analyzer.py", line 377 in plotVoltageDomain
  File "/usr/local/Cellar/python/3.7.5/Frameworks/Python.framework/Versions/3.7/lib/python3.7/tkinter/__init__.py", line 1705 in __call__
  File "/usr/local/Cellar/python/3.7.5/Frameworks/Python.framework/Versions/3.7/lib/python3.7/tkinter/__init__.py", line 1283 in mainloop
  File "/Users/dole/.virtualenvs/py3/lib/python3.7/site-packages/frc_characterization/drive_characterization/data_analyzer.py", line 1312 in main
  File "/Users/dole/.virtualenvs/py3/lib/python3.7/site-packages/frc_characterization/cli/__init__.py", line 41 in analyzerDrive
  File "/Users/dole/.virtualenvs/py3/lib/python3.7/site-packages/frc_characterization/cli/__init__.py", line 90 in main
  File "/Users/dole/.virtualenvs/py3/bin/frc-characterization", line 8 in <module>
Abort trap: 6`

randomstring, 2020-01-19:
I did several things to try and fix this. I reinstalled tcl-tk and python3 with no improvement. Ended up stumbling into this thread about mathplotlib failing on the mac  https://github.com/pyenv/pyenv-virtualenv/issues/140 and I created a ~/.mathplotrc file with the line `backend      : Agg` in it and now frc-characterization is no longer crashing when generating the plots.

pietroglyph, 2020-01-19:
Thank you for reminding me about this! I walked a user through fixing this issue on the frc-discord and made a branch with the fix, and then completely forgot to PR it!

pietroglyph, 2020-01-19:
Please try pulling #89 and installing it locally (`pip install git+https://github.com/pietroglyph/robot-characterization@mojave-fix --upgrade`).

PR #89 (2020-01-20): Fix tk+matplotlib on macOS Mojave
I walked a user through diagnosing this on the FRC Discord about a month ago, and then I forgot to PR the fix! Fortunately #88 reminded me. Please don't merge this until the user who opened that issue confirms that this fix works.

Closes #88.
closed 2020-01-20 by Oblarg