wpilibsuite/frc-characterization #95: Robot Characterization does not specify brushed or brushless

Brandonh2987, 2020-01-22:
Our team tried to do robot characterization with Spark Maxes. It automatically set the motors to brushed and burned half of our drive train out. There is no way to tell if a motor is brushed or brushless with robot characterization without looking at the code itself. Could you please add an option to the robot characterization tool to specify what mode the motors are supposed to be in.

Oblarg, 2020-01-22:
It is intended that users use the `Neo` project for Neo motors.  We'll change the name on the spark max project to `Spark Max (Brushed)` to reduce confusion.

auscompgeek, 2020-03-10:
This is resolved with #97 and #107 now, right?

Starlight220, 2020-09-16:
I think this is resolved.