wpilibsuite/frc-characterization #104: Referring to the SparkMaxes Motor Controllers in the Config Editor using CAN

Tyler4338, 2020-01-25:
[image]

Hi, I am starting to do motion profiling for my team, 4338, and I just didn't know how I would refer to the SparkMaxes in the config editor by their CAN id's. We have a six wheel drive with 6 SparkMaxes and didn't decide to use them through PWM. I assumed the default values when I created a brushless max project are for PWM, so it's for the best that I ask.

prateekma, 2020-01-25:
The Spark MAX projects (both brushed and brushless) interact with the motor controller over the CAN bus, so these numbers refer to the CAN IDs. If you have 3 motors on each side, you can simply add the ID of the third motor to the existing list. Remember that the first number should be the ID of the master motor and the latter two should be the IDs of the slave motors.

matia6170, 2020-01-25:
[image]
I tried doing somthing simliar but it wont generate a project.

sciencewhiz, 2020-01-25:
You also need to add a third item to the inversion settings.