wpilibsuite/frc-characterization #105: WPILib update (2020.2.2) incompatibility

lmkenney, 2020-01-25:
Project will not deploy with WPILib update (2020.2.2) - error message indicates it is looking for the older version of the library

Daltz333, 2020-01-25:
Could you please give the full error message?

This project is standalone to the normal WPILib install so it shouldn't matter.

lmkenney, 2020-01-25:
FAILURE: Build failed with an exception.

* Where:

Build file 'C:\Users\TB-Th\characterization-project\build.gradle' line: 3

* What went wrong:

Plugin [id: 'edu.wpi.first.GradleRIO', version: '2020.1.2'] was not found
in any of the following sources:

- Gradle Core Plugins (plugin is not in 'org.gradle' namespace)

- Plugin Repositories (could not resolve plugin artifact
'edu.wpi.first.GradleRIO:edu.wpi.first.GradleRIO.gradle.plugin:2020.1.2')

  Searched in the following repositories:
    MavenLocal(file:/C:/Users/TB-Th/.m2/repository)
    Gradle Central Plugin Repository
    frcHome(file:/C:/Users/Public/wpilib/2020/maven/)

* Try:

Run with --stacktrace option to get the stack trace. Run with --info or
--debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 1s

sciencewhiz, 2020-01-25:
If you do a deploy while connected to the internet it will download the dependencies, and then you can do another deploy while connected to the robot.

Starlight220, 2020-09-18:
Was this resolved? WPILib is now on 2020.3.2 anyway.