wpilibsuite/frc-characterization #112: PyNetworkTables Errors

adam-mcdaniel, 2020-02-03:
Whenever we try to characterize our robot now, we get this error.
```
NT: connect() to 172.22.11.2 port 1735 timed out
Exception in Tkinter callback
Traceback (most recent call last):
  File "C:\Users\6517\AppData\Local\Programs\Python\Python38\lib\tkinter\__init__.py", line 1883, in __call__
    return self.func(*args)
  File "C:\Users\6517\AppData\Local\Programs\Python\Python38\lib\site-packages\frc_characterization\logger_gui\__init__.py", line 101, in connect
    NetworkTables.addConnectionListener(
TypeError: addConnectionListener(): incompatible function arguments. The following argument types are supported:
    1. (self: _pyntcore._ntcore.NetworkTablesInstance, callback: Callable[[_pyntcore._ntcore.ConnectionNotification], None], immediate_notify: bool) -> int

Invoked with: <_pyntcore._ntcore.NetworkTablesInstance object at 0x000002CCA7512870>, <bound method TestRunner.connectionListener of <frc_characterization.drive_characterization.data_logger.TestRunner object at 0x000002CCA75B3F40>>; kwargs: immediateNotify=True
```

We had characterized our bot several times before, and then this randomly shows up. Our PyFRC is up to date. Is there anything else we can do?

virtuald, 2020-02-03:
This is a pyntcore bug, as it is not 100% compatible with pynetworktables yet (but we are working on getting it there!). https://github.com/robotpy/pyntcore/issues/11 is relevant.

... though, I thought this had been fixed in pyntcore 2020.2.2.0? What version do you have installed?

virtuald, 2020-02-03:
A workaround for now is to create a separate virtual environment, and install frc-characterization in there without pyntcore. However, long term I do intend to fix pyntcore to be compatible with frc-characterization, just need to do it.

adam-mcdaniel, 2020-02-03:
> yet (but we are working on getting it there!). robotpy/pyntcore#11

I have the newest version of PyFRC installed, if that helps. Im away from
the FRC computer right now but Ill let you know as soon as I can.

virtuald, 2020-02-04:
Fixed in pyntcore 2020.2.2.1

closed 2020-02-04 by virtuald