wpilibsuite/frc-characterization #115: Post-Encoder Gearing and Encoder EPR for Spark Max

TeaAndPotatoes, 2020-02-17:
This year, REV added support for external encoders in brushless mode but in the Data Analyzer, the "Post-Encoder Gearing" and "Encoder EPR" text fields (which of course influence the calculated kP and kD) are only editable when the Controller Type is set to "Talon" and not for "Spark".

From looking in the code briefly, it seems like the Talon gain analysis works exactly like the Spark Max's analysis, except with Post-Encoder Gearing and Encoder EPR taken into account. Am I right on this?

If I am, it would be awesome to have the "Spark" Controller Type work with the two aforementioned fields, just like with "Talon", and I'd be happy to create a PR for a small change like this.

auscompgeek, 2020-02-18:
My understanding is that the intended usage is to set the encoder conversion factors on the Spark Max to account for gearing (which always gets multiplied by encoder revolutions - the Spark Max will always account for EPR for you).

TeaAndPotatoes, 2020-02-19:
Ah, I always forget about the conversion factor functionality on the Spark Max. I'm going to mark this closed given that there's no further action needed. Thanks for the response!

closed 2020-02-19 by TeaAndPotatoes