wpilibsuite/frc-characterization #116: Decrease sensor sample rate from 20ms to 5ms

calcmogul, 2020-02-18:
This will better capture acceleration dynamics, resulting in a better fit for `K_a`.

Oblarg, 2020-02-18:
This should be accompanied by minimizing the velocity filtering in the smart motor controller project templates.

pietroglyph, 2020-02-19:
Just a note about the uniform data format: the drive logger runs an extra track width test, and as a result it has an extra set of data for that test. Any additional unification of the data format will need to manage this reality. I might have a different picture of what this effort looks like than you do, but I would like to point out that there are 4 separate logger GUIs and 16 different templates—moving complexity from the loggers to the templates moves the duplication from 4 to 16 places unless we're careful.

closed 2020-10-28 by calcmogul