wpilibsuite/frc-characterization #117: Feed WPI_TalonFX objects or turn off motor safety

qhdwight, 2020-02-19:
See: https://github.com/CrossTheRoadElec/Phoenix-api/issues/68

Or else amber lights on slave will occur and there will be no movement.

Tested on our x4 Falcon drivetrain. Default generated project doesn't work. When motor safety was disabled by manually going into Java code it worked.

auscompgeek, 2020-02-19:
I think the best way to do this is to actually make the followers the base concrete class instead of the WPI_* ones.

pietroglyph, 2020-02-19:
Turning off motor safety is actually what we want in this situation (unless we want to call `feed` on the followers), and explicitly disabling it is a lot clearer than implicitly turning it off via the base non-WPI classes, IMO.

auscompgeek, 2020-02-19:
I suppose for the drivetrain characterisation that would make sense. But for simple-motor for example we don't need them to implement SpeedController.

pietroglyph, 2020-02-19:
That’s a good point. I used the WPI classes when I wrote simple motor in order to be consistent with the other projects (you have to enter in WPI_TalonSRX in the other projects’ configs after all), but we could switch to the base class everywhere (not using DifferentialDrive could be good because of the whole motor inversion misfeature anyway.)

Ultimately I err on the side of lazy, so I would prefer disabling motor safety (and if there are no other serious objections then I’ll go ahead and PR that in the next couple of days.)