wpilibsuite/frc-characterization #124: Automatically compensate for sensor delay in CTRE REV onboard PID

calcmogul, 2020-03-20:
If we know the sensor delay, we can use equation (11) from https://www.textroad.com/pdf/JBASR/J.%20Basic.%20Appl.%20Sci.%20Res.,%202(3)2607-2618,%202012.pdf to correct the controller gain to compensate for the delay. The following figures are for an elevator model running with 5ms updates and 50ms of input lag (input lag is a mathematical dual of measurement lag). The first figure is the LQR without delay compensation.

[image]

The second figure is the LQR with delay compensation.

[image]

The continuous time, infinite horizon LQR is defined as `u = -R^-1 B^T P x`. Based on equation (11), the corrected equation should be `u = -R^-1 B^T P e^(A - B P R^-1 B^T P) x`. (Note that equation (11) has a typo of `P^-1` when they meant `R^-1`.)

For the discrete control law `u_k = -K x_k`, making the controller gain `K e^(A - B P R^-1 B^T P)` instead of `K` gave the second figure above where `A`, `B`, `P`, and `R` are from the continuous model and `P` is from solving the continuous ARE. Here's my python code using frccontrol that's run once after the initial `K` is computed:

```python
        Q = np.diag(1.0 / np.square([0.02, 0.4]))
        R = np.diag(1.0 / np.square([12.0]))
        P = sp.linalg.solve_continuous_are(a=self.sysc.A, b=self.sysc.B, q=Q, r=R)
        self.K = self.K @ sp.linalg.expm((self.sysc.A - self.sysc.B @ np.linalg.pinv(R) @ self.sysc.B.T @ P) * sensor_delay_in_seconds)
```

data_analyzer.py doesn't have access to the type of feedback controller used (CTRE/REV onboard PID vs WPILib PID), so we'll need to refactor things so the tool can selectively apply this controller gain correction automatically.

auscompgeek, 2020-03-20:
To make things more annoying, CTRE allow for configuring the velocity rolling average window.

pietroglyph, 2020-03-20:
We could add an input box for input lag to the analyzer. We should weigh the tradeoffs between configurability and the difficulty users have with more settings. I think adding another input probably isn’t an issue if we provide sane defaults.

calcmogul, 2020-04-16:
This gives the same results as above, so we won't need the CARE.
```python
        Q = np.diag(1.0 / np.square(q))
        R = np.diag(1.0 / np.square(r))
        self.K = self.K @ sp.linalg.expm(
            (self.sysc.A - self.sysc.B @ self.K) * len(self.ubuf) * 0.005
        )
```

PR #125 (2020-07-05): Add LQR latency compensation
This adds an extra measurement delay input box in the data analyzer, along with appropriate defaults for the various motor controller presets. Note that this changes the behavior of the loop type dropdown so that it now applies the presets over whatever the user has when the dropdown is changed; this is to make it so that the measurement delay preset is changed when the user changes the loop type automatically.

Closes #124.
closed 2020-07-05 by Oblarg