wpilibsuite/frc-characterization #138: Only upload to pypi on tagged releases

calcmogul, 2020-10-22:
Not having every commit be a release will give us more freedom in the beta testing process.

pietroglyph, 2020-10-22:
Does this mean that we're going to have frc-characterization-specific tasks in the beta? The CI/CD system we have is a result of this project's historically *very* different approach to releases and testing compared to e.g. allwpilib.

Piphi5, 2020-10-22:
> Does this mean that we're going to have frc-characterization-specific tasks in the beta? The CI/CD system we have is a result of this project's historically *very* different approach to releases and testing compared to e.g. allwpilib.

I believe so. I think the main reason is that they want to merge #126 but it hasn't been fully tested and documented so updating Pypi is probably not the best idea.

PR #139 (2020-10-28): Move to Actions, push to PyPI on tag
Closes #138
closed 2020-10-28 by calcmogul