wpilibsuite/frc-characterization #158: [CRASH] IsADirectoryError: logger_analyzer/templates

RyanHir, 2021-01-09:
Fresh installation on Arch Linux using guide from https://docs.wpilib.org/en/stable/docs/software/wpilib-tools/robot-characterization/introduction.html

Extra package installed from the repos `tk`

When running `frc-characterization drive new` per documentation instructions, I receive this error...

```
Traceback (most recent call last):
  File "/home/ryan/.local/bin/frc-characterization", line 8, in <module>
    sys.exit(main())
  File "/home/ryan/.local/lib/python3.9/site-packages/frc_characterization/cli/__init__.py", line 97, in main
    tool_dict[args.mech_type][args.tool_type](directory=args.project_directory)
  File "/home/ryan/.local/lib/python3.9/site-packages/frc_characterization/cli/__init__.py", line 25, in new_project
    newproject.main(testType)
  File "/home/ryan/.local/lib/python3.9/site-packages/frc_characterization/newproject/__init__.py", line 498, in main
    project.configureGUI()
  File "/home/ryan/.local/lib/python3.9/site-packages/frc_characterization/newproject/__init__.py", line 389, in configureGUI
    updateTemplatePath()
  File "/home/ryan/.local/lib/python3.9/site-packages/frc_characterization/newproject/__init__.py", line 121, in updateTemplatePath
    with resources.path(mech, "templates") as path:
  File "/usr/lib/python3.9/contextlib.py", line 117, in __enter__
    return next(self.gen)
  File "/usr/lib/python3.9/importlib/resources.py", line 175, in _path_from_reader
    opener_reader = reader.open_resource(norm_resource)
  File "<frozen importlib._bootstrap_external>", line 995, in open_resource
IsADirectoryError: [Errno 21] Is a directory: '/home/ryan/.local/lib/python3.9/site-packages/frc_characterization/logger_analyzer/templates'
```

calcmogul, 2021-01-09:
I had that same error on Arch yesterday, and someone else reproduced it on macOS. It doesn't seem to occur on Windows. We have no idea why it happens yet.

RyanHir, 2021-01-09:
Noted, will use a windows machine for the meantime.

PR #160 (2021-01-11): Open config file directly instead of erroneously opening its parent dir
Closes #158.
PR #162 (2021-01-21): Fixes IsADirectoryError from importlib.resources.path()
... by replacing it with a method that takes a directory not a filepath.

* Reimplements importlib.resources.files() locally because it does not exist in
versions prior to Python 3.9.
* Should properly fix #158 in all the locations it occurs.
* This reverts commit 76cd65b777eb2d3eb151d923dee7c297fb552514
commit 56ae2573 (2021-01-21): Fixes IsADirectoryError from importlib.resources.path() (#162)

This replaces importlib.resources.path() with a method that takes a
directory instead of a filepath. It essentially reimplements
importlib.resources.files() because that function doesn't exist in
Python 3.8 and earlier. This fixes #158 in all the locations it occurs.

Reverts commit 76cd65b.
closed 2021-01-11 by calcmogul