wpilibsuite/frc-characterization #159: Program abort on start (macOS 11.1)

prateekma, 2021-01-09:
```
macOS 11 or later required !
[1]    55200 abort      frc-characterization drive new
```

```
Process:               Python [55200]
Path:                  /usr/local/Cellar/python@3.9/3.9.0_5/Frameworks/Python.framework/Versions/3.9/Resources/Python.app/Contents/MacOS/Python
Identifier:            org.python.python
Version:               3.9.0 (3.9.0)
Code Type:             X86-64 (Native)
Parent Process:        zsh [55155]
Responsible:           Terminal [55153]
User ID:               501

Date/Time:             2021-01-09 15:24:34.285 -0500
OS Version:            macOS 11.1 (20C69)
Report Version:        12
Bridge OS Version:     5.1 (18P3030)
Anonymous UUID:        369F0C1A-8DE4-C477-963F-0A5ED2E38DB2

Sleep/Wake UUID:       EE4D69D7-D8D9-4D08-805E-0CEF22A1E55A

Time Awake Since Boot: 230000 seconds
Time Since Wake:       25 seconds

System Integrity Protection: enabled

Crashed Thread:        0  Dispatch queue: com.apple.main-thread

Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000
Exception Note:        EXC_CORPSE_NOTIFY

Application Specific Information:
abort() called

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   libsystem_kernel.dylib        	0x00007fff20358462 __pthread_kill + 10
1   libsystem_pthread.dylib       	0x00007fff20386610 pthread_kill + 263
2   libsystem_c.dylib             	0x00007fff202d9720 abort + 120
3   Tcl                           	0x00007fff6fe57b55 Tcl_PanicVA + 398
4   Tcl                           	0x00007fff6fe57bd5 Tcl_Panic + 128
5   Tk                            	0x00007fff6ff57ad5 TkpInit + 385
6   Tk                            	0x00007fff6fed7788 0x7fff6fea6000 + 202632
7   _tkinter.cpython-39-darwin.so 	0x000000010c6bb701 Tcl_AppInit + 84
8   _tkinter.cpython-39-darwin.so 	0x000000010c6b69fa _tkinter_create + 975
9   org.python.python             	0x000000010b9eecb3 cfunction_vectorcall_FASTCALL + 203
10  org.python.python             	0x000000010ba66b13 call_function + 403
11  org.python.python             	0x000000010ba63ca4 _PyEval_EvalFrameDefault + 27452
12  org.python.python             	0x000000010ba6765b _PyEval_EvalCode + 1998
13  org.python.python             	0x000000010b9bf0ec _PyFunction_Vectorcall + 248
14  org.python.python             	0x000000010b9be97b _PyObject_FastCallDictTstate + 212
15  org.python.python             	0x000000010b9bf3a5 _PyObject_Call_Prepend + 139
16  org.python.python             	0x000000010ba0798e slot_tp_init + 87
17  org.python.python             	0x000000010ba0118b type_call + 150
18  org.python.python             	0x000000010b9beab7 _PyObject_MakeTpCall + 266
19  org.python.python             	0x000000010ba66b47 call_function + 455
20  org.python.python             	0x000000010ba63ca4 _PyEval_EvalFrameDefault + 27452
21  org.python.python             	0x000000010b9bf15c function_code_fastcall + 97
22  org.python.python             	0x000000010b9be97b _PyObject_FastCallDictTstate + 212
23  org.python.python             	0x000000010b9bf3a5 _PyObject_Call_Prepend + 139
24  org.python.python             	0x000000010ba0798e slot_tp_init + 87
25  org.python.python             	0x000000010ba0118b type_call + 150
26  org.python.python             	0x000000010b9beab7 _PyObject_MakeTpCall + 266
27  org.python.python             	0x000000010ba66b47 call_function + 455
28  org.python.python             	0x000000010ba63d50 _PyEval_EvalFrameDefault + 27624
29  org.python.python             	0x000000010b9bf15c function_code_fastcall + 97
30  org.python.python             	0x000000010ba66b13 call_function + 403
31  org.python.python             	0x000000010ba63ca4 _PyEval_EvalFrameDefault + 27452
32  org.python.python             	0x000000010ba6765b _PyEval_EvalCode + 1998
33  org.python.python             	0x000000010b9bf0ec _PyFunction_Vectorcall + 248
34  org.python.python             	0x000000010b9bee33 PyVectorcall_Call + 146
35  org.python.python             	0x000000010bad01f7 partial_call + 137
36  org.python.python             	0x000000010b9beab7 _PyObject_MakeTpCall + 266
37  org.python.python             	0x000000010ba66b13 call_function + 403
38  org.python.python             	0x000000010ba63e0e _PyEval_EvalFrameDefault + 27814
39  org.python.python             	0x000000010b9bf15c function_code_fastcall + 97
40  org.python.python             	0x000000010ba66b13 call_function + 403
41  org.python.python             	0x000000010ba63d50 _PyEval_EvalFrameDefault + 27624
42  org.python.python             	0x000000010ba6765b _PyEval_EvalCode + 1998
43  org.python.python             	0x000000010ba5d08d PyEval_EvalCode + 79
44  org.python.python             	0x000000010ba980f5 run_eval_code_obj + 110
45  org.python.python             	0x000000010ba974ed run_mod + 103
46  org.python.python             	0x000000010ba963b1 PyRun_FileExFlags + 241
47  org.python.python             	0x000000010ba959a1 PyRun_SimpleFileExFlags + 271
48  org.python.python             	0x000000010baad8bd Py_RunMain + 1839
49  org.python.python             	0x000000010baadbf6 pymain_main + 306
50  org.python.python             	0x000000010baadc44 Py_BytesMain + 42
51  libdyld.dylib                 	0x00007fff203a1621 start + 1
```

MudkipMao, 2021-01-19:
Seems to be having issues on Python 3.9, try downgrading to 3.7.

Chief Delphi thread here: https://www.chiefdelphi.com/t/frc-characterization-wont-generate-project/390372/

auscompgeek, 2021-01-19:
@MudkipMao I'm pretty sure that is #158 discussed in that thread. This is a completely different crash.

prateekma, 2021-03-02:
Seems to work on 11.2 without issues now, so something must have been fixed upstream.

closed 2021-03-02 by prateekma