wpilibsuite/frc-characterization #161: What are kG and kCos?

LudumDareDevelopment, 2021-01-10:
Hi,

I was just checking out the tool when I noticed two new constants: kG and kCos. I was wondering if these are the voltage to angular velocity/acceleration constants meant to be passed into `identifyDrivetrainSystem` (for `kVAngular` and `kAAngular` respectively). Also, can the kS kV kA constants provided by the tool still be used for the standard `SimpleMotorFeedforward `calculations?

calcmogul, 2021-01-10:
No, `kG` is for the elevator gravity feedforward term and `kCos` is for the single-jointed arm gravity feedforward. We had those since the 2019 offseason.

To obtain angular constants, there's a checkbox that flips the motor directions so the robot spins in place instead of driving forward. Everything else works the same way.

pietroglyph, 2021-01-10:
kG and kCos are for when you're analyzing an elevator or arm, respectively. The fields for those should be greyed out when you're analyzing a drivetrain.

To get angular constants for a drivetrain, you need to run the data logging routine twice. First run it normally, and then run it again but with the angular mode checkbox checked. You'll need to run both of the resulting data files through the analyzer.

LudumDareDevelopment, 2021-01-10:
Got it, thank you so much. And just to clarify, you would use the linear KS KV KA wrapped in a `SimpleMotorFeedforward` to pass into the Ramsete controller (no state-space/LQR).

pietroglyph, 2021-01-10:
Yes, `RamseteCommand` takes a `SimpleMotorFeedforward` for each of your motors.

(Note that the RAMSETE control law doesn't actually have anything to do with computing feedforwards; `RamseteCommand`, however, handles all of the trajectory following, which includes computing feedforwards.)

LudumDareDevelopment, 2021-01-10:
My apologies `RamseteCommand` was what I was referring to.

closed 2021-01-10 by LudumDareDevelopment