wpilibsuite/frc-characterization #164: Crash when attempting to analyze data

cjdenio, 2021-01-21:
Hello! When I launch frc-characterization in analyzer mode (`frc-characterization drive analyzer`), then click "Select Data File", the program crashes without showing the file selection dialog. I'm on Pop!_OS 20.10 (basically Ubuntu). Thanks!

```bash
❯ frc-characterization drive analyzer
X Error of failed request:  BadLength (poly request too large or internal Xlib length error)
  Major opcode of failed request:  139 (RENDER)
  Minor opcode of failed request:  20 (RenderAddGlyphs)
  Serial number of failed request:  2686
  Current serial number in output stream:  2697
```

cjdenio, 2021-01-21:
Whoops, forgot to send my Python version! I'm running Python 3.8.6

sciencewhiz, 2021-01-26:
Does installing any of the packages here help? https://bugs.launchpad.net/ubuntu/+source/git/+bug/1852985

If you run it from a different directory, does it work (I'm working on the theory that there's something in the directory listing that's unprintable and causes tk to crash).

cjdenio, 2021-01-26:
Just had a "duh" moment: I have some Emojicode (`.🍇`) files in my home dir, and tk must not like rendering emojis :shrug:  So, it works perfectly from another directory. Thanks!

closed 2021-01-26 by cjdenio