wpilibsuite/frc-characterization #166: Track-width computation doesn't account for wheel circumference

chauser, 2021-01-22:
In data_analyzer.py, calcTrackWidth returns the track width in units of wheel revolutions, not the project unit, as can be seen by considering that d_left and d_right are directly computed from table entries that are measured in revolutions in the self.stored_data table.
I noticed this in characterizing a Romi where the calculated track width was about 0.67m. For a more usual robot the scale of the error would be about a factor of 2 (too large, as 6" wheels have about a 0.5m circumference). I wonder if this behind the remark in https://docs.wpilib.org/en/stable/docs/software/wpilib-tools/robot-characterization/characterization-routine.html that  "the empirical trackwidth calculated by frc-characterization may be significantly different from the real trackwidth (e.g., off by a factor of 2)."?

calcmogul, 2021-01-22:
> 6" wheels have about a 0.5m circumference

Six inches corresponds to 0.1524 meters, so what you're saying here doesn't make sense.

chauser, 2021-01-22:
.1524m (diameter) * pi = 0.478m (circumference)
In the pre-processed tables that are used for other computations in the analyzer the raw data in the self.stored_data table are multiplied by self.units_per_rot; that does not seem to be happening for the track width computation. The .json file of characterization data ends with 
```
    "test": "Drivetrain",
    "units": "Rotations",
    "unitsPerRotation": 1.0
}
```
so at least in the data I'm using it is clear that an adjustment needs to be made.

calcmogul, 2021-01-22:
@Piphi5 Looks like we missed this in the unit conversion rewrite.

chauser, 2021-01-22:
I guess a stronger bit of evidence is that regardless of what unit you choose in the data analyzer, the track width output is always the same.

Piphi5, 2021-01-22:
I'll open a PR soon

sciencewhiz, 2021-01-22:
The reason for the note in the docs is because how easily a robot turns may vary quite a bit from the theoretical calculation a lot due to things like wheel type (ie its harder to turn with wide sticky pneumatic wheels).

PR #167 (2021-01-29): Add trackwidth conversions
Still need trackwidth characterization to verify this works.

Should close #166
closed 2021-01-29 by calcmogul