wpilibsuite/frc-characterization #168: TalonFX encoder inversion not working

ValerieBG, 2021-01-23:
Today my team was trying to take a characteristic of our robot using the FRC Characterization tool, but are having issues getting proper encoder readouts.

When we got the project generated using the CTRE template and drove the robot forward, we discovered that our right side drive encoders were reading negative values, so we tried to use the Tool’s function to invert the encoder. It turns out, the change that the tool makes in the code was setSensorPhase() which doesn’t work for the TalonFX per this topic: [Talon .setSensorPhase issues - Programming - Chief Delphi](https://www.chiefdelphi.com/t/talon-setsensorphase-issues/375573/7)

When we found this post, I looked in the code and verified that it is using setSelectedSensorPosition, so there isn’t a problem there, but while digging around in the code, we found that is makes use of the WPILib DifferentialDrive class which automatically inverts the right side drive power.

This function of DifferentialDrive seems to cause an issue where the right side drive will always drive opposite of the internal encoder’s value.

prateekma, 2021-01-23:
Is this actually an issue when analyzing the data? The analyzer takes the absolute value of the encoder measurements and applies the sign of the voltage to compensate for these weird scenarios.

pietroglyph, 2021-01-23:
In addition to what Prateek said, as far as I know you should still be able to get all the inversions right, it's just that the right side might be flipped in a way that it isn't if you don't use `DifferentialDrive`.