wpilibsuite/frc-characterization #174: Can't generate Simple-Motor SparkMax project

zeroClearAmerican, 2021-01-26:
Python 3.9 fresh install. PATH Environment variable includes the correct python directories. Followed all instructions in [the guide](https://docs.wpilib.org/en/latest/docs/software/wpilib-tools/robot-characterization/introduction.html#introduction-to-robot-characterization) to the letter. 

Clicking "Generate Project" fails with the below error. 

Config:
```
{
    # Set to True if the spark max will be brushed
    # ** WARNING: Improperly setting this can break things **
    "brushed": False,
    # Ports for motors
    # If doing drive test, treat this as the left side of the drivetrain
    "motorPorts": [31],
    # Only if you are doing drive (leave empty "[]" if not)
    "rightMotorPorts": [30],
    # Set motors to inverted or not
    # If doing drive test, treat this as the left side of the drivetrain
    "motorsInverted": [true],
    # Only if you are doing drive (leave empty "[]" if not)
    "rightMotorsInverted": [false],
    # Encoder edges-per-revolution (*NOT* cycles per revolution!)
    # **Note pass an EPR of 1 for the NEO Integrated encoder as the SparkMax
    # already handles this conversion**
    # For the REV Through Bore Encoder, use 8192 (2048 * 4)
    "encoderEPR": 1,
    # Gearing accounts for the gearing between the encoder and the output shaft
    "gearing": 1,
    # Set this to True if you would like to use the SparkMax Data Port
    # Note that the Data Port is the 10 pin port on the top of the SparkMax
    # Setting this to False indicates you want to use the 6 pin encoder port
    # which is located in the front of the SparkMax
    "useDataPort": False,
    # Encoder ports (leave empty "[]" if not needed)
    # Specifying encoder ports indicates you want to use Rio-side encoders
    # If doing drive test, treat this as the left side of the drivetrain
    "encoderPorts": [],
    # Only if you are doing drive (leave empty "[]" if not)
    "rightEncoderPorts": [],
    # Set to True if encoders need to be inverted
    # If doing drive test, treat this as the left side of the drivetrain
    "encoderInverted": False,
    # Only if you are doing drive (set to False if not needed)
    "rightEncoderInverted": False,
    # ** The following is only if you are using a gyro for the DriveTrain test**
    # Gyro type (one of "NavX", "Pigeon", "ADXRS450", "AnalogGyro", or "None")
    "gyroType": "None",
    # Whatever you put into the constructor of your gyro
    # Could be:
    # "SPI.Port.kMXP" (MXP SPI port for NavX or ADXRS450),
    # "SerialPort.Port.kMXP" (MXP Serial port for NavX),
    # "I2C.Port.kOnboard" (Onboard I2C port for NavX),
    # "0" (Pigeon CAN ID or AnalogGyro channel),
    # "new WPI_TalonSRX(3)" (Pigeon on a Talon SRX),
    # "" (NavX using default SPI, ADXRS450 using onboard CS0, or no gyro)
    "gyroPort": "",
}
```

Error:
```
Exception in Tkinter callback
Traceback (most recent call last):
  File "c:\program files\python39\lib\tkinter\__init__.py", line 1885, in __call__
    return self.func(*args)
  File "C:\Users\[username]\AppData\Roaming\Python\Python39\site-packages\frc_characterization\newproject\__init__.py", line 142, in genProject
    config = ast.literal_eval(self.config.get())
  File "c:\program files\python39\lib\ast.py", line 105, in literal_eval
    return _convert(node_or_string)
  File "c:\program files\python39\lib\ast.py", line 94, in _convert
    return dict(zip(map(_convert, node.keys),
  File "c:\program files\python39\lib\ast.py", line 85, in _convert
    return list(map(_convert, node.elts))
  File "c:\program files\python39\lib\ast.py", line 104, in _convert
    return _convert_signed_num(node)
  File "c:\program files\python39\lib\ast.py", line 78, in _convert_signed_num
    return _convert_num(node)
  File "c:\program files\python39\lib\ast.py", line 69, in _convert_num
    _raise_malformed_node(node)
  File "c:\program files\python39\lib\ast.py", line 66, in _raise_malformed_node
    raise ValueError(f'malformed node or string: {node!r}')
ValueError: malformed node or string: <ast.Name object at 0x0000022173F69BB0>
```

It looks like this is some malformed config issue, but I double-checked the config and also ran it through a validator to make sure there were no syntax issues and it didn't flag anything. 

Any tips/clues as to what I could be doing wrong?

pietroglyph, 2021-01-26:
`motorsInverted` and `rightMotorsInverted` should be `[True]` and `[False]`.

zeroClearAmerican, 2021-01-26:
Thanks, that got me past the first error. Next issue is that it's throwing out this:
[image]

With this stack trace:
```
2021-01-26 12:58:25,042:042 INFO    : logger              : Config: {'brushed': False, 'motorPorts': [31], 'rightMotorPorts': [30], 'motorsInverted': [True], 'rightMotorsInverted': [False], 'encoderEPR': 1, 'gearing': 1, 'useDataPort': False, 'encoderPorts': [], 'rightEncoderPorts': [], 'encoderInverted': False, 'rightEncoderInverted': False, 'gyroType': 'None', 'gyroPort': '', 'controlType': 'SparkMax', 'controllerTypes': ['CANSparkMax'], 'rightControllerTypes': ['CANSparkMax']}
Exception in Tkinter callback
Traceback (most recent call last):
  File "C:\Users\[username]\AppData\Roaming\Python\Python39\site-packages\frc_characterization\newproject\__init__.py", line 150, in genProject
    with resources.path(res, "project") as path:
  File "c:\program files\python39\lib\contextlib.py", line 117, in __enter__
    return next(self.gen)
  File "c:\program files\python39\lib\importlib\resources.py", line 175, in _path_from_reader
    opener_reader = reader.open_resource(norm_resource)
  File "<frozen importlib._bootstrap_external>", line 995, in open_resource
PermissionError: [Errno 13] Permission denied: 'C:\\Users\\[username]\\AppData\\Roaming\\Python\\Python39\\site-packages\\frc_characterization\\robot\\project'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "c:\program files\python39\lib\tkinter\__init__.py", line 1885, in __call__
    return self.func(*args)
  File "C:\Users\[username]\AppData\Roaming\Python\Python39\site-packages\frc_characterization\newproject\__init__.py", line 177, in genProject
    shutil.rmtree(dst)
  File "c:\program files\python39\lib\shutil.py", line 737, in rmtree
    return _rmtree_unsafe(path, onerror)
  File "c:\program files\python39\lib\shutil.py", line 596, in _rmtree_unsafe
    onerror(os.scandir, path, sys.exc_info())
  File "c:\program files\python39\lib\shutil.py", line 593, in _rmtree_unsafe
    with os.scandir(path) as scandir_it:
FileNotFoundError: [WinError 3] The system cannot find the path specified: 'C:/Users/[username]/Documents/Robotics/flywheel_profiling\\characterization-project'
```

The only file in the project folder is `robotconfig.py`. I've also run the terminal as an administrator to see if that changed the permission

auscompgeek, 2021-01-29:
I believe that last error is the same root cause as #158.

Fdgttygt5, 2021-02-01:
I have had the same issue but for the CTRE control type

Ethan-Sharp, 2021-02-03:
I have this issue as well for In CTRE control type on a windows system.

Piphi5, 2021-02-03:
Try downgrading to Python 3.7

aescorcia21, 2021-02-11:
Has anyone fixed the permission(13, permission denied) I’m getting this error CTRE control type on windows

Fdgttygt5, 2021-02-11:
> Try downgrading to Python 3.7

This fixed the issue for me, thank you.

Ethan-Sharp, 2021-02-13:
This also worked for us.

zeroClearAmerican, 2021-02-13:
I had Python 3.9 and 3.7 installed on the same Windows machine. Uninstalling 3.9, repairing the 3.7 installation, and making sure my environment variables pointed to the right places fixed it for me

closed 2021-02-13 by zeroClearAmerican