wpilibsuite/frc-characterization #176: Min acceleration greater than its max acceleration

alice15311, 2021-01-30:
We've been trying to get our robot to follow a trajectory for a few weeks now without much success. Today, we tried starting over from scratch and re-characterizing the robot. When we finished analyzing the data, we tried to have the robot follow the new trajectory only for it to fail with the error `The constraint's min acceleration is greater than its max acceleration`. We're not using a custom trajectory constraint, so we don't even know where the min acceleration is set. We tried increasing the `kMaxAccelerationMetersPerSecondSquared` constant, but made the error even more ambiguous, only saying something to the effect of "Something went wrong." We're at a loss of what the issue could be, so any help would be greatly appreciated.

[image]

prateekma, 2021-01-30:
Can you paste the code you're using to generate the trajectory?

alice15311, 2021-02-01:
here is the link to the repo we are using 
[https://github.com/Yeti-Robotics/trajectory-testing](https://github.com/Yeti-Robotics/trajectory-testing )