wpilibsuite/frc-characterization #184: Tool does not generate project

viggy96, 2021-02-13:
Tool produces the following error when trying to generate a project:
```
Exception in Tkinter callback
Traceback (most recent call last):
  File "c:\users\mlaze\appdata\local\programs\python\python38-32\lib\tkinter\__init__.py", line 1883, in __call__
    return self.func(*args)
  File "c:\users\mlaze\appdata\local\programs\python\python38-32\lib\site-packages\frc_characterization\newproject\__init__.py", line 142, in genProject
    config = ast.literal_eval(self.config.get())
  File "c:\users\mlaze\appdata\local\programs\python\python38-32\lib\ast.py", line 96, in literal_eval
    return _convert(node_or_string)
  File "c:\users\mlaze\appdata\local\programs\python\python38-32\lib\ast.py", line 85, in _convert
    return dict(zip(map(_convert, node.keys),
  File "c:\users\mlaze\appdata\local\programs\python\python38-32\lib\ast.py", line 81, in _convert
    return list(map(_convert, node.elts))
  File "c:\users\mlaze\appdata\local\programs\python\python38-32\lib\ast.py", line 95, in _convert
    return _convert_signed_num(node)
  File "c:\users\mlaze\appdata\local\programs\python\python38-32\lib\ast.py", line 74, in _convert_signed_num
    return _convert_num(node)
  File "c:\users\mlaze\appdata\local\programs\python\python38-32\lib\ast.py", line 66, in _convert_num
    raise ValueError('malformed node or string: ' + repr(node))
ValueError: malformed node or string: <_ast.Name object at 0x2352BF88>
```

calcmogul, 2021-02-13:
Your robot config file is malformed. Could you post it?

calcmogul, 2021-02-13:
You could also try using Python 3.7.

viggy96, 2021-02-13:
Here you go.
[robotconfig.txt](https://github.com/wpilibsuite/frc-characterization/files/5976277/robotconfig.txt)

Piphi5, 2021-02-13:
There's some missing some controller types and motors inverted. Also, there needs to be quotes around the controllerType names.

For future reference, if you pass in two values for motorPorts, you need to have two values for controllerTypes and motorsInverted.

Here's what your config file should resemble:

```
{
    # Ports for motors
    # If doing drive test, treat this as the left side of the drivetrain
    "motorPorts": [0, 1],
    # Only if you are doing drive (leave empty "[]" if not)
    "rightMotorPorts": [2, 3],
    # Class names of motor controllers used.
    # 'WPI_TalonSRX'
    # 'WPI_VictorSPX'
    # 'WPI_TalonFX'
    # If doing drive test, treat this as the left side of the drivetrain
    "controllerTypes": ['WPI_TalonFX', 'WPI_TalonFX'],
    # Only if you are doing drive (leave empty "[]" if not)
    "rightControllerTypes": ['WPI_TalonFX', 'WPI_TalonFX'],
    # Set motors to inverted or not
    # If doing drive test, treat this as the left side of the drivetrain
    "motorsInverted": [False, False],
    # Only if you are doing drive (leave empty "[]" if not)
    "rightMotorsInverted": [False, False],
    # Encoder edges-per-revolution (*NOT* cycles per revolution!)
    # For the CTRE Mag Encoder, use 16384 (4 * 4096 = 16384)
    "encoderEPR": 16384,
    # Gearing accounts for the gearing between the encoder and the output shaft
    "gearing": 1,
    # Encoder ports (leave empty "[]" if not needed)
    # Specifying encoder ports indicates you want to use Rio-side encoders
    # If doing drive test, treat this as the left side of the drivetrain
    "encoderPorts": [],
    # Only if you are doing drive (leave empty "[]" if not)
    "rightEncoderPorts": [],
    # Set to True if encoders need to be inverted
    # If doing drive test, treat this as the left side of the drivetrain
    "encoderInverted": False,
    # Only if you are doing drive (set to False if not needed)
    "rightEncoderInverted": False,
    # ** The following is only if you are using a gyro for the DriveTrain test**
    # Gyro type (one of "NavX", "Pigeon", "ADXRS450", "AnalogGyro", or "None")
    "gyroType": "NavX",
    # Whatever you put into the constructor of your gyro
    # Could be:
    # "SPI.Port.kMXP" (MXP SPI port for NavX or ADXRS450),
    # "SerialPort.Port.kMXP" (MXP Serial port for NavX),
    # "I2C.Port.kOnboard" (Onboard I2C port for NavX),
    # "0" (Pigeon CAN ID or AnalogGyro channel),
    # "new WPI_TalonSRX(3)" (Pigeon on a Talon SRX),
    # "" (NavX using default SPI, ADXRS450 using onboard CS0, or no gyro)
    "gyroPort": "SerialPort.Port.kMXP",
}
```

randomstring, 2021-02-13:
The EncoderCPR you need to use for Falcon 500s is 8192.  

Here's the characterization project we used for our 4 Falcon 500 drivetrain. Obviously you'll need to fix gearing, Can IDs, and gyro settings.

 https://github.com/FRC-Sonic-Squirrels/2021-characterization-project/blob/main/robotconfig.py

frc-characterization will report one side of your robot driving in the wrong direction. That is "normal."

flamespinner, 2021-02-15:
> The EncoderCPR you need to use for Falcon 500s is 8192.
>
> Here's the characterization project we used for our 4 Falcon 500 drivetrain. Obviously you'll need to fix gearing, Can IDs, and gyro [...]

I was having a similar error to @viggy96 . I tried your config, modified it with the new Falcon 500 IDs, and to use the NavX, however, I'm getting a list index out of range. Most likely due to no Encoder Ports in `"encoderPorts": [],` and `"rightEncoderPorts":[]` . Am I missing a value to specify that I am using Falcon 500s? Is there something else that I'm not doing correctly?

Piphi5, 2021-02-15:
Your config file looks good to me except for the fact that you have a Pigeon specified as the Gyro Type and I think you're using port 43 on the right and left ports. But these wouldn't explain the list error.

Did you fill out the config under the CTRE selector? 

From your reply it seems like you pasted the majority of the config from somewhere else. If you did that under the `Simple` config setup, the generator would want the encoder port fields filled out which isn't your case.