wpilibsuite/frc-characterization #185: Unable to generate project

ThiessenLouisG, 2021-02-18:
I'm trying to make a characterization project for my teams robot. When I try to generate the project I get the error message: Unable to generate project - config might be bad. Details: Permission error (13, 'Permission denied') 

[CLI returns](https://github.com/ThiessenLouisG/6044RobotCode2021/blob/main/err.txt)
[Project file](https://github.com/ThiessenLouisG/6044RobotCode2021/blob/main/robotconfig.py)

pietroglyph, 2021-02-18:
Right now the tool only works with Python 3.7.

ThiessenLouisG, 2021-02-18:
I have python3.7 , but my system wants to run it with 3.9. Does anyone know how to avoid this?

auscompgeek, 2021-02-19:
If you install it using python3.7, it'll run using python3.7.

ThiessenLouisG, 2021-02-22:
I've come to the conclusion that unless you have python 3.7 as your primary python installation this tool is unusable. My primary python installation was 3.9, and even by uninstalling 3.9 and doing a fresh reinstall of 3.7 was unsuccessful. Multiple errors confused the installation and use to the point of python, pip, and the tool being FUBAR. Please extend support to the newest python versions, or provide sufficient documentation on downgrading/upgrading and installation.

closed 2021-02-22 by ThiessenLouisG

Daltz333, 2021-02-22:
We do have Python 3.7 explicitly listed in the installation documentation.

Piphi5, 2021-02-22:
> We do have Python 3.7 explicitly listed in the installation documentation. 

I think the suggestion was to have more explicit documentation on how to downgrade python for those who don't know how to do so.

ThiessenLouisG, 2021-02-23:
> I think the suggestion was to have more explicit documentation on how to downgrade python for those who don't know how to do so.

Exactly. I've tried everything I could do based off hours of research to no avail. If there is a way to downgrade that I am not aware about I greatly encourage the devs to share.

Daltz333, 2021-02-23:
Windows is quite easy. Just run the uninstalled. We can link instructions in documentation, but frc-char doesn't have much lifespan left, as we are creating a C++ equivalent with imgui.