wpilibsuite/frc-characterization #187: Drive Train Characterization Tests Timing Out & Not Recording Data

the1NotHuman, 2021-03-02:
**Version:** 2021.2.2
This did not happen with last year's version.

**Description**: When running any 4 of the drive train characterization tests, at the end of the test when the robot is disabled it throws an error saying the NT timed out and does not collect the data. [Chief Delphi thread](https://www.chiefdelphi.com/t/robot-drive-train-characterization-nt-timeout/391903)

**Expected Behavior:** I expected the tests to complete normally.

**Actual Behavior:** The test throws and error and does not collect the data.

prateekma, 2021-03-02:
Are you using a new project? The project generated by frc-characterization before 2021 will not work.

Piphi5, 2021-03-02:
When you finish a run, is there data in the "telemetry" entry of the smart dashboard? There should be a string sent there. Also if you enable prints it should say how many samples were collected.