wpilibsuite/frc-characterization #195: LLVM: Out of memory

28andrew, 2021-03-28:
This may to relate to or be the cause of #187. When doing a simple motor characterization, after disabling autonomous after running a test (usually the 2nd or third test), the tool doesn't immediately popup reporting amount of rotations like normal and has a popup saying NT timeout a few seconds later instead. It turns out the RoboRIO actually runs out of memory. I recall this happening when I was running each test close to a minute. This is not a problem when I run the tests for a shorter amount of time like 25 seconds.

My suspicion is with the telemetry string on disable:
https://github.com/wpilibsuite/frc-characterization/blob/main/frc_characterization/logger_analyzer/templates/Robot.java.mako#L376-L379

Attached is a driver station log.
[2021_03_27_15_06_09_Sat.zip](https://github.com/wpilibsuite/frc-characterization/files/6218204/2021_03_27_15_06_09_Sat.zip)
I can't get the log messages to show when opening it on another device but one of the last messages is "LLVM: out of memory"

Piphi5, 2021-03-28:
We should probably document that running it for a minute is a bad idea

Modelmat, 2021-04-26:
> 
> 
> We should probably document that running it for a minute is a bad idea

This doesn't particularly make sense. For most systems, characterising 12V is "logical" -> this will take 48s at the default quasistatic ramp rate.

Piphi5, 2021-04-26:
Generally speaking, you don't need to collect data points from 0V to 12V as you only need a change in maybe a few volts (for quasistatic) to get the general trend.  Running the quasistatic tests for too long seems to only increase the likelihood of crashing the roborio (as this issue shows) so this should be documented so that people following a similar logic to yours don't end up with similar issues.

calcmogul, 2021-06-02:
A fix for this has been merged into https://github.com/wpilibsuite/sysid, and SysId will be replacing frc-characterization.

closed 2021-06-02 by calcmogul