wpilibsuite/frc-characterization #196: IndexError upon trying to analyze data

democat3457, 2021-06-11:
I'm getting an IndexError when hitting the Analyze button, which seems to be related to the graph plot - the recent log is below.
```py
2021-06-11 13:13:19.247 Python[38110:4191888] WARNING: <NSOpenPanel: 0x7ff9043b6ed0> running implicitly; please run panels using NSSavePanel rather than NSApplication.
2021-06-11 13:13:22,124:124 INFO    : logger              : Received Data!
2021-06-11 13:13:22,124:124 INFO    : logger              : Converting to rotational measure (fixed conversion)
2021-06-11 13:13:22,164:164 INFO    : logger              : Units: Degrees, Test: Arm, Units per rotation: 360.000
2021-06-11 13:13:22,165:165 INFO    : logger              : Converting to rotational measure (fixed conversion)
Exception in Tkinter callback
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/tkinter/__init__.py", line 1705, in __call__
    return self.func(*args)
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/frc_characterization/logger_analyzer/data_analyzer.py", line 213, in runAnalysis
    test_runners[Tests(self.test.get())]()
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/frc_characterization/logger_analyzer/data_analyzer.py", line 251, in runAnalysisArm
    *self.prepared_data[self.subset.get()], self.test.get()
  File "/Library/Frameworks/Python.framework/Versions/3.7/lib/python3.7/site-packages/frc_characterization/logger_analyzer/data_analyzer.py", line 1485, in calcFit
    cos = np.concatenate((qu[PREPARED_COS_COL], step[PREPARED_COS_COL]))
IndexError: index 5 is out of bounds for axis 0 with size 5
```
The data file is
https://gist.github.com/democat3457/3c627121fee46cea0e284281ea6486c3

The robotconfig.py is
https://gist.github.com/democat3457/1c512ba869dad25ae04cc8f2613faf77

calcmogul, 2021-06-11:
Your data JSON is cut off, so the data can't be processed.

democat3457, 2021-06-12:
oops gist mustve cut it off
[image]
it actually ends here

democat3457, 2021-06-12:
[characterization-data20210611-1312.json.zip](https://github.com/wpilibsuite/frc-characterization/files/6641446/characterization-data20210611-1312.json.zip)
heres the zipped data file

calcmogul, 2021-06-12:
The data displays OK in [SysId](https://github.com/wpilibsuite/sysid), but there's some strange artifacts in it.
[image]
[image]

democat3457, 2021-06-12:
I don't actually know what those mean lol

(This was for an intake on a pivot that goes vertically, and I was trying to characterize it for easy PID.)

calcmogul, 2021-06-12:
Here's what more sane data looks like.
[image]
[image]

calcmogul, 2021-06-12:
I just don't think there's enough clean data here to get a good characterization result. You'll probably have to tune things manually.

As for the frc-characterization crash you saw, I don't foresee that getting fixed because new development has moved to SysId.

democat3457, 2021-06-12:
rip on both counts. if new development is now only going to be on SysId, then shouldn't that be updated in the wpilib docs to reflect that? also, shouldn't this repository be archived then to indicate it's no longer being updated?

Piphi5, 2021-06-12:
SysId hasn't gone through any form of beta testing and still needs vendors to complete certain tasks before it's ready. So frc-char is still the program that should be used. Frc-char will be archived and SysId will be documented sometime before the 2022 season.